meshtastic/web #825: [Bug]: An unhandled error occurred. <Add details here>

klausismegaevent, 2025-09-01:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [ ] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

T-Beam 0.7, Heltec V3

### Connection Type

Serial

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

.

### Operating System

macos

### Browser

chrome

### Expected Behavior

Ich kann auf die Einstellung Seiten nicht mehr zugreifen, um zum Beispiel Bluetooth wieder einzustellen. Sobald ich auf dem MacBook mit Chrome die serielle Schnittstelle hergestellt habe, gibt es andauernd Abbruchfenster

### Actual Behavior

.

### Steps to Reproduce

.

### Relevant console output

```Shell
NotFoundError: Failed to execute 'removeChild' on 'Node': The node to be removed is not a child of this node.
    at nN (https://client.meshtastic.org/index-B-cS7hhl.js:48:90562)
    at go (https://client.meshtastic.org/index-B-cS7hhl.js:48:90013)
    at nN (https://client.meshtastic.org/index-B-cS7hhl.js:48:90863)
    at go (https://client.meshtastic.org/index-B-cS7hhl.js:48:90013)
    at nN (https://client.meshtastic.org/index-B-cS7hhl.js:48:90948)
    at go (https://client.meshtastic.org/index-B-cS7hhl.js:48:90013)
    at nN (https://client.meshtastic.org/index-B-cS7hhl.js:48:90450)
    at go (https://client.meshtastic.org/index-B-cS7hhl.js:48:90013)
    at nN (https://client.meshtastic.org/index-B-cS7hhl.js:48:90948)
    at go (https://client.meshtastic.org/index-B-cS7hhl.js:48:90013)
```

### Screenshots

.

### Additional Context

.

klausismegaevent, 2025-09-01:
.

philon-, 2025-09-10:
Can you please see if you can reproduce the issue using https://client-test.meshtastic.org/ ?

danditomaso, 2025-11-06:
I appreciate that you reported this issue, thank you. Unfortunately, there isn't enough info here to proceed with debugging. Please try  our test environment https://client-test.meshtastic.org and see if your issue is resolved there.

closed 2025-11-06 by danditomaso