meshtastic/web #846: [Bug]: Client.meshtastic.org does not load.

lmead8-jpg, 2025-09-13:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [ ] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

T-Deck

### Connection Type

HTTP

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

2.7.4, currently, also tried 2.6.11 and 2.5.4

### Operating System

Windows 11

### Browser

Chrome 140.0.7339.128 , Edge 140.0.3485.54, Brave

### Expected Behavior

I expected a web page that had tabs that would allow configuration

### Actual Behavior

A web page labeled Meshatstic with  the URL of https://client.meshtastic.org/messages/broadcast/0.  There is a spinning circle that says loading but it never loads.  I have tried three different browsers and three different versions

### Steps to Reproduce

Connect the T-Deck Plus
push the center roll button and turn the unit on
After a few seconds release the button, screen should be blank
go to https://client.meshtastic.org
connect.

### Relevant console output

```Shell

```

### Screenshots

[image]

lmead8-jpg, 2025-09-14:
> @lmead8-jpg <https://github.com/lmead8-jpg> this is always caused by
> having multiple connections to your mesh device. A mesh device can only
> have a single connection at once. Try disconnecting in your IOS/Android
> device and/or close all your browser tabs and try again.

There is only one meshtastic unit attached.  The computer was rebooted
several times.  Between each attempt the browser was shut down and I
confirmed that the device manager showed the T-Deck not connected.

danditomaso, 2025-10-06:
I'm closing this issue as we haven't had any other issues with client.meshtastic.org and I've been using it throughout the day without issue