meshtastic/web #867: [Bug]: Link to Node Location on openstreetmap opens with Notes layer activated

lumagician, 2025-09-27:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists
- [x] I have cleared cache/cookies/storage or tried in a private/incognito window (if applicable)

### Hardware

Not Applicable

### Connection Type

Serial

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

2.6.11.60ec05e

### Operating System

Fedora 42

### Browser

Brave 1.81.137

### Expected Behavior

open location on openstreetmap without explicitely opening the notes layer

### Actual Behavior

 `&layers=N` is appended to the link and therefore the layer is active on page load

### Steps to Reproduce

1. go to the node details vie of a node with known position.
2. click on the link showing the coordinates
3. a new browser tab is being opened, conating the openstreetmap with a marker at the position of the nide.

### Relevant console output

```Shell

```

### Screenshots

[image]

### Additional Context

In my opinion displaying the notes layer really does not make any sense in this context. especially the fact the markers of the notes make it difficulat to see the actual relevant marker (see attached screenshot).

philon-, 2025-09-27:
We should really open the map page and focus on the node marker there

danditomaso, 2025-10-17:
@philon- Good point, I've already added a route for the Map route that accepts a long, lat and zoom, which will allow for a user to click these coordinates and be sent to the Map page (but I am considering if a minimap might be a better UX?)